pioupus/kicadLibCreator #11: design settings. text size, visibility etc

pioupus, 2016-11-30:


Gasman2014, 2016-12-13:
Components have very small default annotation sizes. e.g. a capacitor I placed had its value displayed as 100nF in 0.025mm high text - I though there was a fleck of dust on my screen at first! Also default text font colour seems to be purple - not the default aqua green.

pioupus, 2016-12-14:
Now the 100nF is stored in the Field Display value and not as the value itself. Therefore unfortunatly I can't do anything about the colors. Hopefully the Kicad guys will integrate color settings.

Gasman2014, 2016-12-15:
Shame - although not a major issue - I guess on a different level, identifies on the schematic that the part is already fully defined. 

Still getting very small default text sizes.

pioupus, 2016-12-15:
Have you tried do adjust the design settings in "Extras/Design settings"? You also have to start Eeschema since as far as I know it loads its libraries on start up. (Also something I should ask the Kicad developers)

Gasman2014, 2016-12-15:
How did I not see that before?!

That works - yes, can now adjust settings and position. Still having some issues with this though. For instance the default Capacitor 

```
F0 "C" 25 100 50 H V L CNN
F1 "C" 25 -100 50 H V L CNN
F2 "" 38 -150 50 H V C CNN
```
and the new footprint (with no adjustments of position)
```
F0 "C" 0 0 1 H I L TNN ""
F1 "C0805C104K5RACTU" 0 0 1 H I L TNN ""
F2 "C_0805" 0 0 1 H I L TNN "footprint"
...
F7 "100 nF" 0 0 1 H V L TNN "DisplayValue"
```

and the standard resistor

```
DEF R R 0 0 N Y 1 F N
F0 "R" 80 0 50 V V C CNN
F1 "R" 0 0 50 V V C CNN
F2 "" -70 0 50 V V C CNN
F3 "" 0 0 50 H V C CNN
```

compared to the new version.

```
DEF RC1206JR-076K8L R 0 3 N Y 1 F N
F0 "R" 80 0 40 V V C CNN ""
F1 "RC1206JR-076K8L" 0 0 40 H I L BNN ""
F2 "R_1206" 0 0 40 H I L BNN "footprint"
F3 "/Users/johnpateman/Dropbox/Components/Datasheets/resistor_smd/yageo/rc1206jr-076k8l.pdf" 0 0 40 H I L BNN "datasheet"
```

[image]

New on left and default on right. I am not sure that the orientation and positioning is being respected - I would have thought that the RefDes and Value position and orientation should, by default, be copied from the source device. 

Or have I missed something here?

pioupus, 2016-12-15:
does it work now?

Gasman2014, 2016-12-15:
Yes, that is much better! Works nicely now. Need to try it with a wider range of components to be sure. Like the use of the slightly smaller fonts too.

Gasman2014, 2016-12-17:
Yes, much better:

[image]
New above, old from 'device.lib' below.

Although the LED is slightly strange as it has the cathode and anode marked in the new library version - but when I look at the DRAW parameters the default and the new one seem identical. Not a significant problem but not sure why it is so.

Default (no visible KA marks)

```
# LED
#
DEF LED D 0 40 Y N 1 F N
F0 "D" 0 100 50 H V C CNN
F1 "LED" 0 -100 50 H V C CNN
F2 "" 0 0 50 H V C CNN
F3 "" 0 0 50 H V C CNN
$FPLIST
 LED-3MM
 LED-5MM
 LED-10MM
 LED-0603
 LED-0805
 LED-1206
 LEDV
$ENDFPLIST
DRAW
P 2 0 1 0 -50 50 -50 -50 N
P 3 0 1 0 -80 -25 -125 -65 -120 -40 N
P 3 0 1 0 -65 -40 -110 -80 -105 -55 N
P 3 0 1 0 50 50 -50 0 50 -50 F
X K 1 -200 0 150 R 40 40 1 1 P
X A 2 200 0 150 L 40 40 1 1 P
ENDDRAW
```

And the custom version (K & A marked)

```
DEF SML-LX1206GC-TR D 0 3 N Y 1 F N
F0 "D" 0 100 40 H V C CNN ""
F1 "SML-LX1206GC-TR" 0 0 40 H I L BNN ""
F2 "LED_1206" 0 0 40 H I L BNN "footprint"
F3 "/*****/Components/Datasheets/led/lumex/sml-lx1206gc-tr.pdf" 0 0 40 H I L BNN "datasheet"
F4 "1481841632573" 0 0 40 H I L BNN "key"
F5 "SML-LX1206GC-TR" 0 0 40 H I L BNN "man"
F6 "Lumex" 0 0 40 H I L BNN "manufacturer"
F7 "Green" 0 -100 40 H V C CNN "DisplayValue"
DRAW
P 2 0 1 0 -50 50 -50 -50 N
P 3 0 1 0 -80 -25 -125 -65 -120 -40 N
P 3 0 1 0 -65 -40 -110 -80 -105 -55 N
P 3 0 1 0 50 50 -50 0 50 -50 F
X K 1 -200 0 150 R 40 40 1 1 P
X A 2 200 0 150 L 40 40 1 1 P
ENDDRAW
```

Gasman2014, 2016-12-17:
I can see what it is now - 

Default definition is
DEF LED D 0 40 **Y** **N** 1 F N

Custom is 
DEF SML-LX1206GC-TR D 0 3 **N** **Y** 1 F N

According to the library docs, I see that Parameter 5 is Pin Number Visibility and Parameter 6 is Pin Name Visibility. These seem to be reversed here. Not a fatal error but bit confusing at first.

pioupus, 2016-12-18:
I think this is fixed now. You may have to check your design settings (that "overwrite pin style checkbox)