pioupus/kicadLibCreator #17: links to manufacturer page of part

pioupus, 2016-12-04:


Gasman2014, 2016-12-13:
In Kicad, once a part is imported, selecting the 'Show datasheet' button in the component editor does not work fully as the full path is not included. Is it possible to define the full path using a system path e.g. KISYS_DATASHEETS or something similar (which would improve portability) or can the full path be included in the datasheet location reference?

Gasman2014, 2016-12-15:
This now works - thank you!

One, minor issue - not sure if it can be addressed. I have, for instance, a large collection (168 values) of resistors - all of the same series and therefore the same specification. Each one that I add to the library will download and add a unique version of the 1MB datasheet. Not sure I need a 167MB of redundant data! I guess I could just amend the global template whilst I am adding all the resistors to point to the same document but wonder if it could be an option?

Also is there any possibility of using a KISYS type paths so that this does not have to be hardcoded into the library.

pioupus, 2016-12-15:
there is now under settings the possibility to save the datasheet to path/to/datasheet/md5hash.pdf. So an equal file will have an equal filename. For still beeing able to browse the filesystem there are softlinks created with the original datasheet file name pointing to the md5hash.pdf
I will research about the $KIS_YS variable

Gasman2014, 2016-12-15:
That's a very clever solution!

pioupus, 2016-12-16:
I would also prefer a relative path inside the datasheet field. But as far as I know(at least the latest stable Kicad release) Kicad does not provide a setting for defining a datasheet root path. Do you have any suggestion?

Gasman2014, 2016-12-16:
According to a couple of articles, you should be able to set any environment variable you want within Kicad

https://forum.kicad.info/t/3d-model-path-relative-to-project/3248/2

However, this post seems to suggest that all variables are not actually parsed at the moment and therefore, as an alternative, they use an external config file to set a variable  - don't think this would help in this case though.
https://forum.kicad.info/t/kicad-stepup-step-to-mcad-exporter-latest-version-faster-support-for-env-var-and-alias/3049

Alternatively you put everything relative to your project(s) - and have potentially a lot of repetition/bloat.

https://forum.kicad.info/t/need-advice-for-managing-own-library-repo-project-repo/1760

Suggests the need for a database solution :)