pioupus/kicadLibCreator #19: Make fails

Gasman2014, 2016-12-05:
Qmake completes but Make then fails with

```
Undefined symbols for architecture x86_64:
  "RESTRequest::sslErrors(QNetworkReply*, QList<QSslError> const&)", referenced from:
      RESTRequest::qt_static_metacall(QObject*, QMetaObject::Call, int, void**) in moc_restrequest.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[1]: *** [libkikadlibcreatord.1.0.0.dylib] Error 1
make: *** [sub-src-make_first] Error 2
21:24:47: The process "/usr/bin/make" exited with code 2.
Error while building/deploying project kicadLibCreator (kit: Desktop Qt 5.7.0 clang 64bit)
When executing step "Make"
```

OS X / QT 5.7

Gasman2014, 2016-12-05:
What is `libkikadlibcreatord.1.0.0.dylib` ? I can't find it (or anything like it) in any tree - is it spelt correctly? (Also note that there is a spelling mistake in the project name - presume it should be `KicadLibCreator` not `KicadLibCretor`

pioupus, 2016-12-06:
spellling error fixed.
`libkikadlibcreatord.1.0.0.dylib` should be created by src.pro. you will find this file in build/debug/src

pioupus, 2016-12-06:
thank you for testing by the way!

pioupus, 2016-12-06:
just took out the ssl error code. does it compile now on your tool chain?

Gasman2014, 2016-12-06:
Successful (if slightly tricky) compilation of OS X needing a QT5 install with homebrew, (this does not add make by default and the install needs to be forced), adding LD_LIBRARY and CMAKE paths to my bash_profile. Qmake followed by Make produced a set of .o object files but these refused to link with gcc/clang. Opened project in QT Creator and built from there. Will keep you posted!

Gasman2014, 2016-12-06:
Firstly, I would like to say a big thank you for writing this as it looks really promising. Hopefully, I can make some progress on getting it running fully.

I can now compile and run this within QT Creator. I can enter the API code and library/module paths (OSX helpfully puts these in`/Library/Application Support/....` Do I need to escape the space? I have tried both with and without a \ but no luck.

**Working for me so far:**
Octopart lookup! (although you can't follow the link). Column sizes seem a bit random. Very fast though! Have to input exact part number e.g. Atmega328P will NOT find Atmega328P-PU.
"Rules" menu - brings up a new window with a selected set of `global` rules. I will need to figure out how to use this!

**Not working for me:**
On a couple of occasions I could click on the `KiCad input library` and `KiCad lib creation` tabs to open these panes but I they were both blank and I couldn't see any data.  I relaunched and now the tabs don't do anything at all. 
I can't launch the app from the compiled app package - traceback attached. I think there is some inconsistency with how you have spelt libkicadlibcreator - which in some places is libkikadlibcreator. Perhaps this is at least part of the issue?

Process:               kicadlibcreator [22823]
Path:                  /Users/USER/*/kicadlibcreator.app/Contents/MacOS/kicadlibcreator
Identifier:            home.kicadlibcreator
Version:               0
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           kicadlibcreator [22823]
User ID:               502

Date/Time:             2016-12-06 23:31:13.246 +0000
OS Version:            Mac OS X 10.12.1 (16B2555)
Report Version:        12
Anonymous UUID:        ECF5F27D-6818-A5A1-25FF-823E5D3D9568

Sleep/Wake UUID:       1445082C-4521-4936-8944-1ABD3F605A63

Time Awake Since Boot: 70000 seconds
Time Since Wake:       7900 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    DYLD, [0x1] Library missing

Application Specific Information:
dyld: launch, loading dependent libraries

Dyld Error Message:
  Library not loaded: libkicadlibcreator.1.dylib
  Referenced from: /Users/USER/*/kicadlibcreator.app/Contents/MacOS/kicadlibcreator
  Reason: image not found

Binary Images:
       0x10e724000 -        0x10e727fff +home.kicadlibcreator (0) <F0599BF5-F365-3F7F-A338-B0BE8292740E> /Users/USER/*/kicadlibcreator.app/Contents/MacOS/kicadlibcreator
       0x11d847000 -        0x11d884287  dyld (421.2) <13A9466A-2576-3ABB-AD9D-D6BC16439B8F> /usr/lib/dyld

pioupus, 2016-12-07:
> Do I need to escape the space?

I think it should be ok. but there is a message now if qt cant find the path

> Octopart lookup! (although you can't follow the link).
 
double click. but now a simple click on cell with "link" works aswell.

> Have to input exact part number e.g. Atmega328P will NOT find Atmega328P-PU.

For me I thought I would prefer exact matches. But can be switched to vague in settings now

> "Rules" menu - brings up a new window with a selected set of global rules. I will need to figure out how to use this!

will write more about this.

> On a couple of occasions I could click on the KiCad input library and KiCad lib creation tabs to open these panes but I they were both blank and I couldn't see any data. I relaunched and now the tabs don't do anything at all.

There is a message now why.

> I can't launch the app from the compiled app package

yes.. there is a linking issue. You can start the app by using the start_kicadlibcreator.sh This file sets the library search path. In linux you have to set LD_LIBRARY_PATH. I don't know how it works with MacOS.

But glad that it works so far!

pioupus, 2016-12-07:
Have a look at https://github.com/pioupus/kicadLibCreator/blob/master/docs/howto.md

Gasman2014, 2016-12-08:
Thank you - the Howto is very helpful - I will have a careful read of the rules section.

> Do I need to escape the space?
> I think it should be ok. but there is a message now if qt cant find the path

FYI - I don't think I need escape the space as the app identifies the validity of the path.

> double click. but now a simple click on cell with "link" works swell.

I discovered the double click by accident but the new functionality is better.

Like to option of 'Exact or 'Fuzzy' matching - might be nice to be able to chose between the two on a case by case basis rather than having it in the preferences but having the option is good.

Biggest problem however is I am still not able to access the `KiCad input library` or `KiCad lib creation` tabs. 
Scenario - I do an Octopart search for C0805C104K5RACTU which returns 3 options with 'Exact Match' selected. I identify the first  component and highlight the entire row. When I now click on the `KiCad input library` tab I get a error notification in the bottom corner "Please select a part from Octopart query first"

My paths are valid and do not prompt an error 

Footprints are `/Library/Application Support/kicad/modules`
Libraries are `/Library/Application Support/kicad/library`
Target path `/Library/Application Support/kicad/library2` (which is an empty directory).

The modules contain a host of "Component_XXXX.pretty" directories and the Libraries have a variety of "xxx.lib" files in them (all from the KiCad git repo)

Gasman2014, 2016-12-08:
I am going to close the `make` issue as that is now resolved, thank you very much.
However, I am going to open an issue regarding the tab selection to make matters clearer.

Good progress!

closed 2016-12-08 by Gasman2014