pioupus/kicadLibCreator #30: KiCad input library and KiCad lib creation tabs not working

Gasman2014, 2016-12-08:
Search for a part in `Octopart` tab. e.g. 2N7000BU -> returns one  component.
Select the one component line. 
Clicking on the `KiCad input library` tab prompts the user to `Please select a part from octopart query first`.

pioupus, 2016-12-08:
For me it works very well. I can proceed with a double click, just with a click on the next tab(ok because just one component found) and I can select the line and proceed.

If you can proceed or not is determined via the octopart mpn. If this is empty you can proceed. In line 230 of mainwindow.cpp the program decides whether you can proceed or not.

pioupus, 2016-12-09:
Can you try again and have a look at the commandline outputs? Now it prints out which mpn is actually selected. If it is an empty string you can't proceed. (except if your result is just one line)

Gasman2014, 2016-12-09:
Hiya

Have been having a play with he debugger today as this is still not working, sorry. When I highlight a single row in the Octopart search (with either an exact or fuzzy match) and then click on the KiCad input library tab nothing happens. Terminal output now returns `mpn set to ""`.

I have set up a Virtual Ubuntu 16 box in  and compiled your app on that too with **exactly** the same problem. I thought it might just have been an OS X thing.

Something funny happens around line 235.

Gasman2014, 2016-12-09:
By the way, think the Fuzzy vs Strict search is a good idea and I like it next to the search box. Wonder if it might be a better interface option if you had 'Fuzzy' as the default and just a checkbox next to the search for 'Exact match only'? Could remove it rom prefs then too. Just an idea.

pioupus, 2016-12-09:
thats strange.. now I get prints like:

```mpn set to ""
mpn set to ""
mpn set to ""
row selected: 0 queryResults.count(): 2
mpn set to "2N7000BU"
mpn set to ""
row selected: 0 queryResults.count(): 2
mpn set to "2N7000BU"
query result count:  2
query result[ 0 ] = "query result: mpn 2N7000BU, manufacturer Fairchild Semiconductor desc: N-Channel 60 V 5 Ohm Through Hole Advanced Small Signal Mosfet - TO-92 footprint TO-92"
query result[ 1 ] = "query result: mpn 2N7000TA, manufacturer Fairchild Semiconductor desc: 2N7000BU Series 60 V 5 Ohm N-Channel Advanced Small Signal MOSFET - TO-92 footprint TO-92"
current query result:  "query result: mpn 2N7000BU, manufacturer Fairchild Semiconductor desc: N-Channel 60 V 5 Ohm Through Hole Advanced Small Signal Mosfet - TO-92 footprint TO-92" ```

there are more debug prints now. What toolchain do you use? GCC? Which Qt Version?

Gasman2014, 2016-12-10:
OK, some progress, I think - although this is still not working quite correctly.

I am compiling this all from within the latest Qt Creator (4.0.2 Based on Qt 5.7.0 (Clang 7.0 (Apple), 64 bit). Built on Jun 13 2016). This seems to compile and build the whole project in one pass without any errors. I had tried using gcc / g++ but although a lot of object files were created I had problems linking them. 

If I now do a search which returns only one item e.g. the 2N7000BU mention previously and I highlight the line and click on `KiCad input library` I am prompted to 'select an item first'.

**However** if I do a search for the same item (which returns the same one row) and DO NOT highlight it but instead click straight on `KiCad input library`,  I am taken to the next screen. 

Just to make matters more confusing, if I click back on the Octopart tab and now highlight the single row and pass the `KiCad input library` button, I get taken trough to the next screen.

However, I believe the part number should be displayed in the top part of the `KiCad input library` screen - in none of these scenarios is the active part name being displayed.

Here is part of the transcript;

query result[ 0 ] = "query result: mpn 2N7000BU, manufacturer Fairchild Semiconductor desc: N-Channel 60 V 5 Ohm Through Hole Advanced Small Signal Mosfet - TO-92 footprint TO-92"
current query result:  "query result: mpn , manufacturer  desc:  footprint "
query result count:  1
query result[ 0 ] = "query result: mpn 2N7000BU, manufacturer Fairchild Semiconductor desc: N-Channel 60 V 5 Ohm Through Hole Advanced Small Signal Mosfet - TO-92 footprint TO-92"
current query result:  "query result: mpn , manufacturer  desc:  footprint "
query result count:  1
query result[ 0 ] = "query result: mpn 2N7000BU, manufacturer Fairchild Semiconductor desc: N-Channel 60 V 5 Ohm Through Hole Advanced Small Signal Mosfet - TO-92 footprint TO-92"
current query result:  "query result: mpn , manufacturer  desc:  footprint "
mpn set to ""
proceed because just one line found.
mpn set to "2N7000BU"
query result count:  1
query result[ 0 ] = "query result: mpn 2N7000BU, manufacturer Fairchild Semiconductor desc: N-Channel 60 V 5 Ohm Through Hole Advanced Small Signal Mosfet - TO-92 footprint TO-92"
current query result:  "query result: mpn 2N7000BU, manufacturer Fairchild Semiconductor desc: N-Channel 60 V 5 Ohm Through Hole Advanced Small Signal Mosfet - TO-92 footprint TO-92"

pioupus, 2016-12-10:
ok, let me guess: if you select a row and press enter then it works? http://stackoverflow.com/questions/11816272/qtablewidget-cellactivated-signal-not-working#comment15709805_11817133

Should be the same issue as described in the link because in your debug prints is no line like "row selected: 0 queryResults.count(): 2"

I changed the signal now to `itemSelectionChanged`. Hopefully it works now.

Gasman2014, 2016-12-10:
Yessssss! That works now. Thank you - sorry for all the hassle!
Still a few issues - random quits etc but can make more progress now. Will keep playing with this.

commit 709e49b5 (2016-12-10): closes #30 - less debug prints
closed 2016-12-10 by Gasman2014