pioupus/kicadLibCreator #31: 'Variables' button does not provide possible options

Gasman2014, 2016-12-10:
I am trying to add rules and I have some (like value, mpn, manufacturer etc) working. When i click on the 'Variables' button, I would expect to see a list of possible options (as per the ReadMe)

> For each field there is a text box which you can fill with text or variables. For knowing which variables exist, there is a variable button besides the text box. A window with a list of variable-value pairs appears.  

This does not appear to work for me. For instance, I would like to add the ID and the pop up window shows %rule.name% and the name of the rule and two blank boxes.

[image]

So it has been pretty much guesswork as to what might work :)

Might I suggest that the global rule come ready populated with ID, Datasheet, MPN and manufacturer? These would seem pretty standard and if anyone didn't want them it would be easy to edit.

Also, the rules do not survive a crash :(  - this can be triggered by clicking on the 'Source Device - 'Add' button' for instance.

pioupus, 2016-12-10:
You have to open the rules window from the context of the 3rd tab. Because only then the program knows the specifications of the part. You have to click on the edit button besides the rulename combobox of the 3rd tab.

Gasman2014, 2016-12-10:
Got it. 

Hesitate to ask but I have 3 other questions;

How do I add a datasheet - can't find any reference to it in the possible completion options?

When I press 'create', no file is saved.

In the second screen (KiCad input library) there  is no reference in the top bar area despite selecting the appropriate library.
[image]

pioupus, 2016-12-11:
Dont hesitate to ask! It is quite important that somebody tests the program :-) 

> 1 datasheet:

The path you specify in the datasheet field is the target path where the datasheet is downloaded to. I would recommend something like `%rule.name%/%octo.manufacturer.saveFilename%/%octo.mpn.saveFilename%.pdf`. When you click on download or create the datasheet should be downloaded in that folder relative to the datasheet path in the settings menu. 

> 2 creation:

A library with the name specified in the bottom combox should be created in the folder "target path"  of the settings window. The path and file should be created automatically. The part will have the name as the def/f1 field value. I have not tested what happens if the fields you mention are empty.

> 3, link in in input tab:

The reason why you don't see any link in the top part of the window is that you have no rule defined yet which A is linked to the octopart category of the queried octopart part and B has no source device defined.
To do this: select an octopart part, a symbol and enter the kicad lib creation tab. There you have to clikc on the edit button. In the rules window you find the category and source device tab with proposed categories/ source devices. A double click on the proposed item will link the rule to the category or device.

Thank you by the way for finding the crash which happens when you click on the source device's add button when no proposed device is selected. If you observe more crashes please tell me.

pioupus, 2016-12-11:
by the way: there is an example rule file in the root path which might help you: `partCreationRules.ini` copy this file in the execution directory of the lib creator.

Gasman2014, 2016-12-11:
Thank you. Sorry I am not more able to help by contributing to your code but hopefully my questions will help address the problems a newbie might have!

1. Datasheet
Thank you - that works. In fact it had worked before but i hadn't noticed the new directory pop up in my datasheet folder. However, when you click on 'Show' I would expect to either open the data sheet or at least shown it in the directory explorer - neither of which happens.

2. Library creation
Unfortunately, even when all of the fields are populated, when I click 'Create' nothing happens. I have tried both with and without a pre-existing *.lib file.

3. Link in 'Input tab'.
Strange problem here. When I go back to this tab, I notice that the list of possible libraries has been duplicated. If I click on the copy of the library lower in the list the application exits.

BW

closed 2016-12-13 by pioupus