pioupus/kicadLibCreator #33: Library creation: nothing happens

pioupus, 2016-12-12:
Unfortunately, even when all of the fields are populated, when I click 'Create' nothing happens. I have tried both with and without a pre-existing *.lib file.

pioupus, 2016-12-12:
show message in status bar whether file could be created or not

pioupus, 2016-12-12:
there should be a message now which explains what happens. how does it work now?

Gasman2014, 2016-12-13:
Build problems - I think you haven't included `ui_mainwindow.h` ?

pioupus, 2016-12-13:
This file is created by qt. Just run qmake again. click on build clean and after that on build all

Gasman2014, 2016-12-13:
OK, works now!!Thank you - looking really promising. Now just to batch convert a chunk of my Partkeepr MySQL database of components :)

 Minor issue regarding paths. Default location for the Kicad libraries in OS X is `/Library/Application Support/kicad/library`. Trying to save to (an existing directory) library2 in the same location fails, I think, due to the space in the filename as I checked the directory permissions. I have been saving elsewhere without issue.

pioupus, 2016-12-13:
Great glad to hear this! So you think this program is usable now?
I would check the permission again because i think its still a permission issue.

Gasman2014, 2016-12-13:
Yes, definitely think this is a functional solution & really useful solution - well done. I will check I can launch it from the script as up to now I have been running it within QT Creator. My only comments on improvement relate around the interface design which, although very functional might benefit from the odd adjustment here and there

Rechecked permissions of library2 - looks OK to me.
```
drwxr-xr-x  161 johnpateman  admin    5474 Oct 10 16:03 library
drwxrwxrwx    4 johnpateman  admin     136 Dec 10 21:10 library2
```
The save message at the bottom of the screen says 

> Could not save library "/library/application_support/kicad/library2/test2.lib"
which led me to wonder if it might be the space or possibly the capitalisation.

closed 2016-12-14 by pioupus