pioupus/kicadLibCreator #39: Spice directives

Gasman2014, 2016-12-15:
On the same basis as you have defined other component values, could you pull any values into the Spice simulator (i.e. by defining Spice_Primitive & Spice_Model?) For passives you only need R, C or I and a value. Would need more work for more complex models but they could be stored like data sheets are currently.

pioupus, 2016-12-15:
This sounds interesting. Do you use the Kicad schematics in your spice simulations?

Gasman2014, 2016-12-15:
Since the more recent versions of Kicad have included the simulator, yes. Helpful for simulating at least parts of the schematic. Not sure if the simulator is included in the release version but it has been available in the nightlys for quite a while.