pioupus/kicadLibCreator #41: integrate db lib workflow

pioupus, 2016-12-16:
In this workflow normal database tables are used as source device rather than Kicad symbols. Then also users who use database libs (like Altium etc) can use this tool.

Gasman2014, 2016-12-16:
I am using PartKeepr (https://partkeepr.org) which uses either a mySql or a Postgresql backend. Would be good to incorporate your lib creator with this - Octopart integration has been mentioned by several users as a feature request but progress seems quite slow - it is quite a professional looking product though.
Just link in qeda ( https://github.com/qeda/qeda)  for complete component and IPC7351C footprint generation and you are done!

Gasman2014, 2016-12-18:
Also Kicadlibrarian (http://www.compuphase.com/electronics/kicadlibrarian_en.htm) will connect to a MySQL or PostgreSQL repository.

pioupus, 2016-12-19:
Thank you for the links! they look very interesting! Especially because I still look for a source for IPC compliant footprints.

Gasman2014, 2016-12-19:
If you want some IPC footprints, try here - https://github.com/cpavlina/kicad-pcblib
These are IPC7351B versions (I think) - vertical format, not all 'round rectangle' pads & not sure if all the courtyards and refdes are on the correct layers.

Gasman2014, 2017-01-06:
Also have had a provisional play with KicadPartManager (http://mikecrash.com/index.php?name=Content&pa=showpage&pid=10) which also runs on a MySQL backend. This has quite a few rough edges and still needs a bit of work but might integrate better with KicadLibCreator (if you aim is to build an Octopart validated  custom component database). It has the ability to track component stock and to parse a schematic as a CvPCB replacement. I am having some problems getting to to run fully however (maybe an OS X Issue :) )

Gasman2014, 2017-07-15:
Any progress on this? It would be a good alternative to a 'batch processing' workflow.