pioupus/kicadLibCreator #48: Batch mode

Gasman2014, 2017-07-15:
Any possibility of allowing a 'batch mode'. ie. by submitting a file containing a list of part numbers that can be read in and processed? I have a long lost of resistors and capacitors that I would like to generate libraries from - cut and paste is a little tedious.

pioupus, 2017-07-18:
Unfortunately I am too busy to implement this feature right now.

What should happen if there are more than one Octopart entries and the result is ambiguous? Or if the rules dont describe all fields sufficiently?

Gasman2014, 2017-07-20:
I was thinking that it could be a kind of semi-auto mode. You could read in a file of component names (I am thinking here of say an E series of resistors) and pop up the Octopart query for the first component in the list  allowing you to proceed to select the correct item and click from there. When you have added the component then to go back to the next component in the list automatically bringing up the Octopart search pane.

 Just to avoid having to type in each number in turn (or having to cut 'n paste between two applications.

Just an idea :)

John